aws4embeddedlinux/meta-aws #361: Dunfell branch does not build due to new override syntax being used

MartinLoeper, 2022-06-23:
Hi!
I am not sure whether I misunderstand something about the Yocto versioning here...

It seems like it is not possible to build the dunfell branch starting from the following commit:
https://github.com/aws4embeddedlinux/meta-aws/commit/6d4bd338b7fa2c535f84c1be1efbc841fe64fd25

It is because some recipes contain [overrides with the new syntax](https://docs.yoctoproject.org/migration-guides/migration-3.4.html#override-syntax-changes). Is it possible to somehow use those recipes on dunfell? Am I missing something?

rpcme, 2022-06-26:
depends on the version on Dunfell you're using. After last August, Dunfell was made to use both old and new syntax. Also when doing so ensure to change the local.conf setting for conf version to 2.

MartinLoeper, 2022-06-26:
Thanks for the info @rpcme. I knew I was missing something! Will update our dunfell version! Thanks!!

closed 2022-06-26 by MartinLoeper