aws4embeddedlinux/meta-aws #411: Question: Why does Corretto rdepend on X11?

MartinLoeper, 2022-07-11:
I am currently building greengrass-bin on kirkstone and it complains that distro feature X11 is disabled.
I checked and confirmed we do not need X11 at all for our distro.

I also confirmed that X11 is needed by corretto since https://github.com/aws4embeddedlinux/meta-aws/commit/8fd10095751b56647105e4c9a0074845785cb2af. Is there any chance to work around this runtime dependency? It feels like a really heavy dependency for an IoT device...

thomas-roos, 2022-07-12:
Hi, did spend some time on investigation, but don't have a full nice solution yet - some of this must also be discussed internally.
The root cause of this problem is that when you build your image with `PACKAGE_CLASSES ?= "package_rpm"`, which is the default, then the `dnf` tool which is used in the image creation phase will throw an error message.
```
 Problem: conflicting requests                                                                                                                                                                                     
  - nothing provides libGL.so.1()(64bit) needed by corretto-8-bin-8.332.08.1-r0.core2_64                                                                                                                           
  - nothing provides libX11.so.6()(64bit) needed by corretto-8-bin-8.332.08.1-r0.core2_64                                                                                                                          
  ...
```
This is because the corretto-*-bin packages contain also binaries, which are linked against those missing libs.
If you use `package_ipk` there are no such checks.
In this case the recipe could be modified with something like this:
`RDEPENDS:${PN} += " \
    ${@bb.utils.contains('DISTRO_FEATURES', 'alsa', 'libasound', '', d)} \
    ${@bb.utils.contains('DISTRO_FEATURES', 'x11', 'cairo pango gtk+ libgl libxi libxtst', '', d)} \
"`
But what's the correct solution for `package_rpm` is under investigation - thank you for this report.

MartinLoeper, 2022-07-13:
I see. Thanks for looking into it and confirming @thomas-roos!

I enabled X11 distro feature and it works now.
However, it almost doubled our rootfs size. This is definitely a workaround :(

thomas-roos, 2022-07-15:
good news - I think I found a good solution for now: https://github.com/aws4embeddedlinux/meta-aws/commit/2ca105f6f4adb4e5fbe02db6908f6d33d958062a
please try (make modifications manually) and give feedback.
Will take some time till it's in kirkstone branch, but knowing it's a working solution it might speed up the process.

MartinLoeper, 2022-07-17:
I tested it on kirkstone and it seems to work well. Great job!!!! Thanks a lot!   
I will report if I observe any issue.

closed 2022-07-17 by MartinLoeper