aws4embeddedlinux/meta-aws #420: Zeus Branch

anamod, 2022-07-12:
Hi!
English is not my first language and I was wondering what it means "Breakingfix only" near the Zeus release branch.
Cause I tried to add meta-aws in a recipes in my yocto (Vs Zeus 2.3) and it says it's not compatible.

rpcme, 2022-07-12:
Hello, did you check out the zeus branch after cloning? LAYER_COMPAT on zeus branch is zeus, and you should only receive this error if LAYER_COMPAT is kirkstone - which could mean that zeus branch is not checked out and you're still on mainline.

anamod, 2022-07-13:
@rpcme I checked the LAYER_COMPAT, and it's actually kirkstone (as u said). But what can I do to made it compatible w/ Zeus. 

> which could mean that zeus branch is not checked out and you're still on mainline

I dont get how I can resolve this (sorry but I'm totally new to this stuff and it's pretty confusing)

thomas-roos, 2022-07-13:
Hi,
no problem - just do an `git checkout zeus` in the meta-aws folder.
Zeus is already [end of life](https://wiki.yoctoproject.org/wiki/Releases) - so we recommend using a more recent version like kirkstone. This is also the reason for "breakfix only".
Basically this is all you need to start building:
```
mkdir yoctotest
cd yoctotest/
git clone git://git.yoctoproject.org/poky -b  kirkstone
git clone https://github.com/aws4embeddedlinux/meta-aws -b kirkstone
git clone https://github.com/openembedded/meta-openembedded.git -b kirkstone
source poky/oe-init-build-env
bitbake-layers add-layer ../meta-openembedded/meta-oe
bitbake-layers add-layer ../meta-openembedded/meta-python
bitbake-layers add-layer ../meta-openembedded/meta-networking
bitbake-layers add-layer ../meta-aws
```
strongly recommending start reading manuals here:  https://docs.yoctoproject.org/overview-manual/concepts.html#layers

anamod, 2022-07-13:
I read only that Zeus is EOL but unfortunately I have to use it; it's for work and we are using hardware that only support Zeus 2.3

thomas-roos, 2022-07-14:
Just let us know if you get stuck - we are here to help you!

closed 2022-07-14 by thomas-roos