aws4embeddedlinux/meta-aws #426: Fatal Error while running pubsub.py

monishkapadia, 2022-07-14:
Hi,

I'm trying to run the `pubsub.py` demo from sdk-python-v2 (1.7.0) on my kirkstone yocto build but I'm facing with this error:

```
****:/conf/iot-sdk# python3 pubsub.py --endpoint ************.iot.us-east-1.amazonaws.com \
>  --root-ca /conf/iot-sdk/certs/AmazonRootCA1.pem \
>  --cert /conf/iot-sdk/certs/certificate.pem.crt \
>  --key /conf/iot-sdk/certs/private.pem.key 
Connecting to ************-ats.iot.us-east-1.amazonaws.com with client ID 'test-*******-****-4c46-967b-3aebbd96f6aa'...
Fatal error condition occurred in /usr/src/debug/aws-c-mqtt/0.7.6-r0/git/aws-c-mqtt/source/packets.c:158: username.len > 0
Exiting Application
################################################################################
Stack trace:
################################################################################
Segmentation fault
```

monishkapadia, 2022-07-15:
I did a rebuild with the master branch of meta-aws and I'm getting a new error. Trying to debug it out, I think somehow the aws-mqtt-c module did not link with the python during compilation, it definitely built all the C modules.

```
Traceback (most recent call last):                                                                                                                                                                              
  File "/conf/iot-sdk/pubsub.py", line 4, in <module>                                                                                                                                                           
    from awscrt import mqtt                                                                                                                                                                                     
  File "/usr/lib/python3.10/site-packages/awscrt/mqtt.py", line 9, in <module>                                                                                                                                  
    import _awscrt                                                                                                                                                                                              
ImportError: /usr/lib/python3.10/site-packages/_awscrt.cpython-310-arm-linux-gnueabi.so: undefined symbol: aws_checksums_crc32  
```

thomas-roos, 2022-07-19:
Could reproduce your error, found that my previous added patch has broken the correct linking of the python c binding. This should be fixed by https://github.com/aws4embeddedlinux/meta-aws/pull/443
Please confirm.

monishkapadia, 2022-07-19:
I'm back to the same error as before, Segmentation fault. I have my system read-only, is that the reason it is faulting up?

monishkapadia, 2022-07-20:
Got the fix for this, thanks to Thomas Roos, Noah Beard and the team. 
To get this working, add `username` parameter to the function `mqtt_connection = mqtt_connection_builder.mtls_from_path` in command_line_utils.py

```
mqtt_connection = mqtt_connection_builder.mtls_from_path(
            endpoint=self.get_command_required(self.m_cmd_endpoint),
            port=self.get_command_required("port"),
            cert_filepath=self.get_command_required(self.m_cmd_cert_file),
            pri_key_filepath=self.get_command_required(self.m_cmd_key_file),
            ca_filepath=self.get_command(self.m_cmd_ca_file),
            on_connection_interrupted=on_connection_interrupted,
            on_connection_resumed=on_connection_resumed,
            client_id=self.get_command_required("client_id"),
            username="some_username",
            clean_session=False,
            keep_alive_secs=30,
            http_proxy_options=proxy_options)
```

sbSteveK, 2022-07-21:
Alternately, you can try adding the argument '--enable_metrics_collection false' if you would like to not set a username at all. There is a fix in the works to prevent the need for either of these temporary fixes.