aws4embeddedlinux/meta-aws #473: Avoiding AUTOREV

paulacarrillo, 2022-07-27:
https://github.com/aws4embeddedlinux/meta-aws/blob/dd742bafa5eff33ac930e97dde851f41cbde85e8/recipes-sdk/aws-iot-device-sdk-python-v1/aws-iot-device-sdk-python-v1.inc#L10

Hi, is it possible to get rid of SRCREV = "${AUTOREV}"? as this could be problematic and goes out of sync?. If not, what is the reason of use it?. We got some LIC_FILES_CHKSUM issues with a customer building our TI's PSDKLA. This commit fix the issue, but we think it could arise again.

Similar request was addressed here: https://github.com/aws/meta-aws/commit/5f0583370dfa92fd33beed910a9135f0f6e1c098#diff-81127c0b77167293e857938fd298c121f013e29e9357e6a8a27a4990c2af2519

Thanks for your help.
Paula Carrillo

thomas-roos, 2022-07-27:
Hi,
you are absolutely right - we have removed this in [master](https://github.com/aws4embeddedlinux/meta-aws/blob/master/recipes-sdk/aws-iot-device-sdk-python-v1/aws-iot-device-sdk-python-v1_1.5.2.bb) and backporting those recipes to kirkstone and dunfell the next weeks. You are using dunfell release?

paulacarrillo, 2022-07-27:
Hi Thomas, thanks for your reply. Yes, we are using dunfell. Do you know the timeline? just want to see if this changed aligns with our next PSDK release =).

thomas-roos, 2022-07-27:
When will this be?

paulacarrillo, 2022-07-27:
Hi Thomas, code freeze is in about 1 week
thanks
Paula

thomas-roos, 2022-07-28:
Hi Paula,
we decided to speed up the backporting process for dunfell. Is there any other recipe you need? Or just all ;)
Best, 
Thomas

paulacarrillo, 2022-07-28:
Just all =) to be sure. Appreciate your help.
Paula

thomas-roos, 2022-08-03:
Hi Paula,
dunfell is up-to-date - if there is anything you need, let us know.

Cheers, 
Thomas

closed 2022-08-03 by thomas-roos