aws4embeddedlinux/meta-aws #575: aws-sdk-cpp do_configure fails

mangodan2003, 2022-09-09:
```
| -- S2N found target: AWS::crypto
| -- crypto Include Dir: /home/dan/yocto/build/tmp/work/core2-64-linux/aws-sdk-cpp/1.9.308-r0/recipe-sysroot/usr/include
| -- S2N found target: crypto
| -- crypto Include Dir: /home/dan/yocto/build/tmp/work/core2-64-linux/aws-sdk-cpp/1.9.308-r0/recipe-sysroot/usr/include
| CMake Error at /home/dan/yocto/build/tmp/work/core2-64-linux/aws-sdk-cpp/1.9.308-r0/recipe-sysroot/usr/lib/s2n/cmake/modules/FindLibCrypto.cmake:21 (get_target_property):
|   get_target_property() called with non-existent target "crypto".
| Call Stack (most recent call first):
|   /home/dan/yocto/build/tmp/work/core2-64-linux/aws-sdk-cpp/1.9.308-r0/recipe-sysroot-native/usr/share/cmake-3.22/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
|   /home/dan/yocto/build/tmp/work/core2-64-linux/aws-sdk-cpp/1.9.308-r0/recipe-sysroot/usr/lib/s2n/cmake/s2n-config.cmake:9 (find_dependency)
|   /home/dan/yocto/build/tmp/work/core2-64-linux/aws-sdk-cpp/1.9.308-r0/recipe-sysroot-native/usr/share/cmake-3.22/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
|   /home/dan/yocto/build/tmp/work/core2-64-linux/aws-sdk-cpp/1.9.308-r0/recipe-sysroot/usr/lib/aws-c-io/cmake/aws-c-io-config.cmake:4 (find_dependency)
|   /home/dan/yocto/build/tmp/work/core2-64-linux/aws-sdk-cpp/1.9.308-r0/recipe-sysroot-native/usr/share/cmake-3.22/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
|   /home/dan/yocto/build/tmp/work/core2-64-linux/aws-sdk-cpp/1.9.308-r0/recipe-sysroot/usr/lib/aws-c-http/cmake/aws-c-http-config.cmake:3 (find_dependency)
|   /home/dan/yocto/build/tmp/work/core2-64-linux/aws-sdk-cpp/1.9.308-r0/recipe-sysroot-native/usr/share/cmake-3.22/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
|   /home/dan/yocto/build/tmp/work/core2-64-linux/aws-sdk-cpp/1.9.308-r0/recipe-sysroot/usr/lib/aws-crt-cpp/cmake/aws-crt-cpp-config.cmake:3 (find_dependency)
|   cmake/AwsFindPackage.cmake:19 (find_package)
|   aws-cpp-sdk-core/CMakeLists.txt:522 (aws_use_package)
```

thomas-roos, 2022-09-09:
Hi, 
could you please provide more details about your setup - e.g. what branch are you using?

Cheers, 
Thomas

mangodan2003, 2022-09-09:
sorry yes, that was a bit vague.

branch is kirkstone current latest as of today, though currently not the very latest for poky and meta-openembedded due to a bug in the most recent release of gstreamer. 
I have tried with a more recent (about a month ago) version of these two layers but the problem persists.

thomas-roos, 2022-09-09:
just tried with the exact same versions - just works.
Do you have something else added?
 - might you can disable other layers if there are some?
 - try to change MACHINE to qemux86-64  eg. run `MACHINE=qemux86-64 bitbake core-image-minimal`

thomas-roos, 2022-09-09:
Another things to do are:
- try to clean cache, sstate-cache
- provide us your BSP and MACHINE config

mangodan2003, 2022-09-13:
Yes we have various other layers. knowing it should work I'll try clearing the caches to start with and if that doesn't help try a  minimal build without other layers to work out where the problem is coming from.
Thanks

mangodan2003, 2022-09-13:
Sure enough removing some other layers has resolved the problem, I'll post back when Iv'e figure out which, but might be  while as not an immediate priority atm.

thomas-roos, 2022-09-13:
great to hear - please let us know when you have more details.

mangodan2003, 2022-09-13:
heh, it appears that `meta-iot-cloud`  I was using for google services was the culprit, and that it also contains some aws recipes and maybe is leading to the problem.

closed 2022-09-13 by thomas-roos