aws4embeddedlinux/meta-aws #585: aws-sdk-cpp submodules

mangodan2003, 2022-09-13:
I want to use the polly voice module, it's included with` aws-sdk-cpp` source but there seems to be no way to enable building it in the existing recipe. Is there a recommended way to  achieve this or should I just roll my own .bbappend?

thomas-roos, 2022-09-14:
hi, there are plans to make that easier for customers. By now I just modified the recipe to build it by default. Should be in master by monday - have a look: https://github.com/aws4embeddedlinux/meta-aws/pull/587
Let me know if that works for you.

mangodan2003, 2022-09-14:
ok thanks, in the mean time I have got by with a bbappend file in one of our local layers which makes those changes.

thomas-roos, 2022-09-14:
Yes👍

closed 2022-09-14 by thomas-roos