aws4embeddedlinux/meta-aws #1070: No match for argument: python3-setuptools Error: Unable to find a match

jahanzaib1234, 2022-11-07:
i get this error

repo: using cache for: oe-repo
not found other for: 
not found modules for: 
not found deltainfo for: 
not found updateinfo for: 
oe-repo: using metadata from Mon 07 Nov 2022 04:58:03 PM UTC.
Last metadata expiration check: 0:00:04 ago on Mon 07 Nov 2022 04:58:04 PM UTC.
No module defaults found
No match for argument: python3-setuptools
Error: Unable to find a match

and my python3-setuptools_65.5.1.bb is 

SUMMARY = "Easily download, build, install, upgrade, and uninstall Python packages"
HOMEPAGE = "https://github.com/pypa/setuptools"
AUTHOR = "Python Packaging Authority <distutils-sig@python.org>"
LICENSE = "MIT License (MIT)"
LIC_FILES_CHKSUM = "file://LICENSE;md5=7a7126e068206290f3fe9f8d6c713ea6"

SRC_URI = "https://files.pythonhosted.org/packages/c5/41/247814d8b7a044717164c74080725a6c8f3d2b5fc82b34bd825b617df663/setuptools-65.5.0.tar.gz"
SRC_URI[md5sum] = "9018a18990e7c16159ec3972affba76c"
SRC_URI[sha256sum] = "512e5536220e38146176efb833d4a62aa726b7bbff82cfbc8ba9eaa3996e0b17"

S = "${WORKDIR}/setuptools-65.5.0"

RDEPENDS_${PN} = ""

thomas-roos, 2022-11-07:
Hi, 
what is the branch you are working on?
What are you trying to build?

jahanzaib1234, 2022-11-07:
dunfell

jahanzaib1234, 2022-11-07:
i am trying to build image for raspberry pi4 and want to to auto boot device with PyQt5 application

jahanzaib1234, 2022-11-07:
for PyQt5 application i need screeninfo and some other modules available in below list 

 /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-screeninfo \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-apscheduler \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-tzlocal \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-pytz \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-sseclient \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-gcloud \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-firebase-admin \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-requests \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-charset-normalizer \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-idna \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-urllib3 \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-certifi \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-google-api-python-client \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-httplib2 \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-google-auth \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-cachetools \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-pyasn1-modules \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-pyasn1 \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-google-auth-httplib2 \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-google-api-core \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-googleapis-common-protos \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-protobuf \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-uritemplate \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-google-cloud-storage \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-google-cloud-core \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-google-resumable-media \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-google-crc32c \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-pyjwt \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-onglass \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-setuptools \
  "

jahanzaib1234, 2022-11-07:
when i try 
IMAGE_INSTALL_append = " python3-screeninfo"

i get this error 

No match for argument: python3-screeninfo
Error: Unable to find a match

After some research i get this point to remove this error i add  "inherit setuptools"  and i get this error 

Could not inherit file classes/setuptools.bbclass

to remove this error i try to install steptools but get same error for steptools

jahanzaib1234, 2022-11-07:
@thomas-roos  and my local.conf file looks like this 

#
# This file is your local configuration file and is where all local user settings
# are placed. The comments in this file give some guide to the options a new user
# to the system might want to change but pretty much any configuration option can
# be set in this file. More adventurous users can look at local.conf.extended
# which contains other examples of configuration which can be placed in this file
# but new users likely won't need any of them initially.
#
# Lines starting with the '#' character are commented out and in some cases the
# default values are provided as comments to show people example syntax. Enabling
# the option is a question of removing the # character and making any change to the
# variable as required.

#
# Machine Selection
#
# You need to select a specific machine to target the build with. There are a selection
# of emulated machines available which can boot and run in the QEMU emulator:
#
#MACHINE ?= "qemuarm"
#MACHINE ?= "qemuarm64"
#MACHINE ?= "qemumips"
#MACHINE ?= "qemumips64"
#MACHINE ?= "qemuppc"
#MACHINE ?= "qemux86"
#MACHINE ?= "qemux86-64"
#
# There are also the following hardware board target machines included for 
# demonstration purposes:
#
#MACHINE ?= "beaglebone-yocto"
#MACHINE ?= "genericx86"
#MACHINE ?= "genericx86-64"
#MACHINE ?= "edgerouter"
#
# This sets the default machine to be qemux86-64 if no other machine is selected:
MACHINE ??= "raspberrypi4"
LICENSE_FLAGS_WHITELIST="commercial"
IMAGE_INSTALL_append = " util-linux-agetty"
IMAGE_INSTALL_append = " python3-tkinter"
IMAGE_INSTALL_append = " python3-pip"
IMAGE_INSTALL_append = " python3-pyqt5"
IMAGE_INSTALL_append = " python3-pyqt5"
IMAGE_INSTALL_append = " python3-pycrypto"
IMAGE_INSTALL_append = " python3-iniparse"
IMAGE_INSTALL_append = " python3-cryptography"
IMAGE_INSTALL_append = " python3-six"
IMAGE_INSTALL_append = " python3-setuptools"
IMAGE_INSTALL_append = " python3-click"
IMAGE_INSTALL_append = " python3-cachecontrol" 
IMAGE_INSTALL_append = " python3-screeninfo" 
IMAGE_FSTYPES = "tar.xz ext3 rpi-sdimg"

# Where to place downloads
#
# During a first build the system will download many different source code tarballs
# from various upstream projects. This can take a while, particularly if your network
# connection is slow. These are all stored in DL_DIR. When wiping and rebuilding you
# can preserve this directory to speed up this part of subsequent builds. This directory
# is safe to share between multiple builds on the same machine too.
#
# The default is a downloads directory under TOPDIR which is the build directory.
#
#DL_DIR ?= "${TOPDIR}/downloads"

#
# Where to place shared-state files
#
[... 203 lines cut ...]
#
# A shared hash equivalent server can be set with "<HOSTNAME>:<PORT>" format
#
#BB_HASHSERVE = "auto"
#BB_SIGNATURE_HANDLER = "OEEquivHash"

# CONF_VERSION is increased each time build/conf/ changes incompatibly and is used to
# track the version of this file when it was generated. This can safely be ignored if
# this doesn't mean anything to you.
CONF_VERSION = "1"

## systemd settings
DISTRO_FEATURES:append = " systemd"
VIRTUAL-RUNTIME:init_manager = "systemd"
VIRTUAL-RUNTIME:initscripts = ""
IMX_DEFAULT_DISTRO_FEATURES:append = " systemd"

## enable some hardware
ENABLE_I2C = "1"
ENABLE_UART = "1"
DISABLE_SPLASH = "1"
DISABLE_RPI_BOOT_LOGO = "1"

IMAGE_INSTALL:append = " psplash"
IMAGE_FEATURES += " splash "

thomas-roos, 2022-11-07:
please share your bblayers.conf and your bitbake command

jahanzaib1234, 2022-11-08:
@thomas-roos 

bitbake command : 

source ./oe-init-build-env smartGlass/build

bitbake qt5-image

 My   bblayers.conf  file 

# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  ${TOPDIR}/../../meta \
  ${TOPDIR}/../../meta-poky \
  ${TOPDIR}/../../meta-yocto-bsp \
  ${TOPDIR}/../../../meta-raspberrypi \
  ${TOPDIR}/../../../meta-openembedded/meta-oe \
  ${TOPDIR}/../../../meta-openembedded/meta-multimedia \
  ${TOPDIR}/../../../meta-openembedded/meta-networking \
  ${TOPDIR}/../../../meta-openembedded/meta-python \
  ${TOPDIR}/../../../meta-openembedded/meta-perl \
  ${TOPDIR}/../../../meta-qt5 \
  ${TOPDIR}/../../../meta-security \
  ${TOPDIR}/../meta-rpi \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-screeninfo \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-apscheduler \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-tzlocal \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-pytz \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-sseclient \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-gcloud \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-firebase-admin \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-requests \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-charset-normalizer \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-idna \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-urllib3 \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-certifi \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-google-api-python-client \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-httplib2 \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-google-auth \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-cachetools \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-pyasn1-modules \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-pyasn1 \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-google-auth-httplib2 \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-google-api-core \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-googleapis-common-protos \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-protobuf \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-uritemplate \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-google-cloud-storage \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-google-cloud-core \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-google-resumable-media \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-google-crc32c \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-pyjwt \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-onglass \
  /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-setuptools \
  "

jahanzaib1234, 2022-11-08:
and setuptools.bbclass looks like this :

inherit distutils

DEPENDS += "python-setuptools-native"

jahanzaib1234, 2022-11-08:
When i remove " inherit setuptools " from " python3-screeninfo_0.8.1.bb " then i get this error 

ERROR: qt5-image-1.0-r0 do_rootfs: No manifest generated from: python3-setuptools

jahanzaib1234, 2022-11-08:
and also remove from local.conf

jahanzaib1234, 2022-11-08:
i am not using meta-aws

jahanzaib1234, 2022-11-08:
ERROR: ParseError at /home/smart/Desktop/smart_glass/poky/smartGlass/sources/meta-screeninfo/recipes-devtools/python/python3-screeninfo_0.8.1.bb:16: Could not inherit file classes/setuptools.bbclass

jahanzaib1234, 2022-11-08:
this is the main error that we need to handle

jahanzaib1234, 2022-11-08:
how can i inherit      "inherit file classes/setuptools.bbclass"

thomas-roos, 2022-11-08:
Hi, sorry to say this, but this is all about meta-aws. May ask your questions here: https://www.yoctoproject.org/community/mailing-lists/ or if the problem is Raspberry Pi related here: https://github.com/agherzan/meta-raspberrypi
Btw. setuptools does not exsist in dunfell use setuptools3 -> https://git.yoctoproject.org/poky/tree/meta/classes/setuptools3.bbclass?h=dunfell
good luck!

closed 2022-11-08 by thomas-roos

jahanzaib1234, 2022-11-08:
@thomas-roos thanks for you help i get the point i need to use setuptools3