aws4embeddedlinux/meta-aws #1495: Space Usage for `botocore` python library

caiotoledo-lunasystems, 2022-12-05:
I'm using [amazon-cloudwatch-publisher](https://github.com/awslabs/amazon-cloudwatch-publisher) to publish journal logs and metrics to `CloudWatch` in one of our embedded projects.

But I've noticed that `botocore` python library takes a lot of space in our system:
```
# du -sh /usr/lib/python3.10/site-packages/boto*
1.4M    /usr/lib/python3.10/site-packages/boto3
44.0K   /usr/lib/python3.10/site-packages/boto3-1.26.20.dist-info
83.6M   /usr/lib/python3.10/site-packages/botocore
240.0K  /usr/lib/python3.10/site-packages/botocore-1.29.21.dist-info
```

Most part of these `83.6 MB` is from its data folder.

Looking at `amazon-cloudwatch-publisher` code, it doesn't use all libraries from `botocore`.

Would it be possible to split `botocore` into multiple packages so the applications which depend on it could use a subset of `botocore` data files? Therefore using less disk space.

thomas-roos, 2022-12-05:
Hi, thank you for this - will investigate and keep you updated.

caiotoledo-lunasystems, 2022-12-07:
Seems that there are discussions about reducing `botocore` data folder size here: https://github.com/boto/botocore/issues/2365
In case this feature is implemented this issue will be solved.

thomas-roos, 2022-12-08:
I think this will be solved when that feature is implemented. Hope this is done soon.

closed 2022-12-08 by thomas-roos