aws4embeddedlinux/meta-aws #2887: AWS-LC conflicts with Openssl

ghensto, 2023-03-11:
I saw the following error while building my sdk using Yocto kirkstone 

```
file /usr/lib/libcrypto.so conflicts between attempted installs of aws-lc-1.5.0-r0.cortexa9t2hf_neon and openssl-dev-3.0.8-r0.cortexa9t2hf_neon
  file /usr/lib/libssl.so conflicts between attempted installs of aws-lc-1.5.0-r0.cortexa9t2hf_neon and openssl-dev-3.0.8-r0.cortexa9t2hf_neon
```

I believe the problem is related to the change of installed path form "${libdir}/aws-lc/*" to "${libdir}/*". Aws-lc tries to replace existing shared libs installed by Openssl

thomas-roos, 2023-03-11:
Hi, 
are you installing aws-lc explicit or through a dependency?
aws-lc and openssl will not work together since aws-lc is a openssl fork and will build same name shared libs.
This was broken in the past by installing to a subdir aws-lc, but they were not used afaik. It just did use the installed openssl, which is not good to build against a different lib than the used one.

ghensto, 2023-03-20:
@thomas-roos my apologizes for the late reply. 

aws-lc and openssl are being installed explicitly in our image's recipe. In our Yocto configuration Qt 5 libraries install OpenSSL 3 as a requirement. My thought was to remove aws-lc but I'm not sure about the consequences of the change being made in your fork of openssl and if that should overwrite the default ssl shared libraries. Also, couple of weeks ago, we had some trouble using the openssl crypto libraries within our aws-iot setup and had to use the one provided by aws-lc .

thomas-roos, 2023-03-21:
Thanks for the response!
What package of your "aws-iot setup" is depending on aws-lc?
What were the troubles?
Afaik aws-lc is not necessary, because all aws-iot libs can also use openssl.

thomas-roos, 2023-03-27:
Will convert this to a discussion - probably more customers will have this question