aws4embeddedlinux/meta-aws #4061: Error in installing awsiot

nealdeore, 2023-06-02:
I am trying to install awsiot on yocto build for Rpi4b

I added below line to local.conf - 
IMAGE_INSTALL_append = "aws-iot-device-sdk-python-v2 python3-pip"

I am facing below error - 
ERROR: Nothing RPROVIDES 'aws-iot-device-sdk-python-v2' (but /home/...../poky/meta/recipes-core/images/core-image-minimal.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'aws-iot-device-sdk-python-v2' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['aws-iot-device-sdk-python-v2']
ERROR: Required build target 'core-image-minimal' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-minimal', 'aws-iot-device-sdk-python-v2']

I am new to yocto. Any help will be much appreciated. Thank you

nateglims, 2023-06-02:
Hi Neal,

Were you able to add the `meta-aws` layer to your `bblayers.conf` file?

It also appears that you are on an older release. If you have nothing else tying you to that release, we recommend using a newer one such as Kirkstone.

nealdeore, 2023-06-06:
Hi Nate,

I do not have a meta-aws layer. this python3-aws-iot-device-sdk-python_1.4.8.bb file is under meta-openembedded/meta-python/recipes-devtool/python.

nateglims, 2023-06-06:
Ah, I see. I think the issue is that they do not have the -v2 recipe. You should add `python3-aws-iot-device-sdk-python` to your image or add our layer and use the v2 recipe.

nealdeore, 2023-06-07:
could you please guide me how can i add your layer to use the v2 recipe. Apologies if this is a dumb question but I am new to this and trying to learn.

nateglims, 2023-06-07:
Sure. If you are using Poky you can use [bitbake-layers](https://docs.yoctoproject.org/4.2.1/singleindex.html#adding-a-layer-using-the-bitbake-layers-script) after activating your build environment to add the dependency layers and meta-aws to your layer config.

```
git clone -b $RELEASE https://git.yoctoproject.org/poky
git clone -b $RELEASE https://git.openembedded.org/meta-openembedded
git clone -b $RELEASE https://github.com/aws4embeddedlinux/meta-aws.git

. ./poky/oe-init-build-env build

bitbake-layers add-layer ../meta-openembedded/meta-oe
bitbake-layers add-layer ../meta-openembedded/meta-python
bitbake-layers add-layer ../meta-openembedded/meta-networking
bitbake-layers add-layer ../meta-openembedded/meta-multimedia
bitbake-layers add-layer ../meta-aws
```

Where $RELEASE Is the yocto release you are using.

Your `conf/bblayers.conf` file should then have a variable `BBLAYERS` that lists all the layers.

Then add to your config (`conf/local.conf`):

(dunfell or later)
```
IMAGE_INSTALL:append = "aws-iot-device-sdk-python-v2"
```

Before dunfell:
```
IMAGE_INSTALL_append = "aws-iot-device-sdk-python-v2"
```

We do recommend using more recent releases if possible.

nateglims, 2023-06-29:
Let us know if you need any more help

closed 2023-06-29 by nateglims