aws4embeddedlinux/meta-aws #4407: aws-iot-device-sdk-python-v2 (kirkstone)

eflavio, 2023-06-30:
Hi,

I want to build aws-iot-device-sdk-python-v2 in yocto kirkstone. However, the build fails at the dependency aws-c-sdkutils. 
This is the bitbake output when I want to build aws-iot-device-sdk-python-v2:

`Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-phytec-linux-gnueabi"
MACHINE              = "ksp0671-1"
DISTRO               = "ampliphy"
DISTRO_VERSION       = "BSP-Yocto-Ampliphy-i.MX6-PD22.1.0"
TUNE_FEATURES        = "arm vfp cortexa9 neon thumb callconvention-hard"
TARGET_FPU           = "hard"
workspace            = "kirkstone:ac518546dc60b7e74bfe3b4234e6af52daecda93"
meta                 
meta-poky            = "HEAD:c8423297866441055553b378663b5454e63e18b4"
meta-oe              
meta-networking      
meta-python          
meta-multimedia      
meta-filesystems     
meta-perl            = "HEAD:50d4a8d2a983a68383ef1ffec2c8e21adf0c1a79"
meta-qt5             = "HEAD:44d44933200287f7d17cf6981af4b4a0961c308d"
meta-rauc            = "HEAD:e2a60ded9c08118c17c23aa07cca21df77fb6628"
meta-security        
meta-tpm             = "HEAD:c79262a30bd385f5dbb009ef8704a1a01644528e"
meta-qt5-phytec      = "HEAD:bd9ac458bac01b899308ac173444607a10f6cc43"
meta-ampliphy        = "HEAD:3c37615e96fb5fe0230abc6a09773e7b89250179"
meta-phytec          = "HEAD:ffcd0a44c8868b9af645e6122ff3ddfb53b2db0f"
meta-rqm             = "HEAD:4a7d79c71ed8a3fffc59b470e73c5c5d75c85fcb"
meta-ksp0671         = "HEAD:2c60d05394016264c176dffc6ff764efa2c269da"
meta-aws             = "kirkstone-next:4c2b6ddaf880f1025a6a7b6d77e0b7647d86b8d2"
meta-iot-cloud       = "kirkstone:65e2c5b2e5cb33760b2deb0fc249ab43ebdf08ef"

Initialising tasks: 100% |##################################################################################################################################################################| Time: 0:00:03
Sstate summary: Wanted 62 Local 9 Mirrors 0 Missed 53 Current 2819 (14% match, 98% complete)
NOTE: Executing Tasks
ERROR: aws-c-sdkutils-0.1.11-r0 do_compile: ExecutionError('/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/temp/run.do_compile.425191', 1, None, None)
ERROR: Logfile of failure stored in: /home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/temp/log.do_compile.425191
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: VERBOSE=1 cmake --build /home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/build --target all --
| [1/11] /home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/recipe-sysroot-native/usr/bin/arm-phytec-linux-gnueabi/arm-phytec-linux-gnueabi-gcc -DAWS_SDKUTILS_EXPORTS -DAWS_SDKUTILS_USE_IMPORT_EXPORT -DHAVE_SYSCONF -Daws_c_sdkutils_EXPORTS -I/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/git/include -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0=/usr/src/debug/aws-c-sdkutils/0.1.11-r0                      -fdebug-prefix-map=/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0=/usr/src/debug/aws-c-sdkutils/0.1.11-r0                      -fdebug-prefix-map=/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/recipe-sysroot-native=  -Wl,-Bsymbolic -DNDEBUG -fPIC -Wall -Werror -Wstrict-prototypes -fno-omit-frame-pointer -pedantic -Wno-long-long -fPIC -fvisibility=hidden -std=gnu99 -MD -MT CMakeFiles/aws-c-sdkutils.dir/source/endpoints_standard_lib.c.o -MF CMakeFiles/aws-c-sdkutils.dir/source/endpoints_standard_lib.c.o.d -o CMakeFiles/aws-c-sdkutils.dir/source/endpoints_standard_lib.c.o -c /home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/git/source/endpoints_standard_lib.c
| FAILED: CMakeFiles/aws-c-sdkutils.dir/source/endpoints_standard_lib.c.o
| /home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/recipe-sysroot-native/usr/bin/arm-phytec-linux-gnueabi/arm-phytec-linux-gnueabi-gcc -DAWS_SDKUTILS_EXPORTS -DAWS_SDKUTILS_USE_IMPORT_EXPORT -DHAVE_SYSCONF -Daws_c_sdkutils_EXPORTS -I/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/git/include -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0=/usr/src/debug/aws-c-sdkutils/0.1.11-r0                      -fdebug-prefix-map=/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0=/usr/src/debug/aws-c-sdkutils/0.1.11-r0                      -fdebug-prefix-map=/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/recipe-sysroot-native=  -Wl,-Bsymbolic -DNDEBUG -fPIC -Wall -Werror -Wstrict-prototypes -fno-omit-frame-pointer -pedantic -Wno-long-long -fPIC -fvisibility=hidden -std=gnu99 -MD -MT CMakeFiles/aws-c-sdkutils.dir/source/endpoints_standard_lib.c.o -MF CMakeFiles/aws-c-sdkutils.dir/source/endpoints_standard_lib.c.o.d -o CMakeFiles/aws-c-sdkutils.dir/source/endpoints_standard_lib.c.o -c /home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/git/source/endpoints_standard_lib.c
| /home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/git/source/endpoints_standard_lib.c:6:10: fatal error: aws/common/json.h: No such file or directory
|     6 | #include <aws/common/json.h>
|       |          ^~~~~~~~~~~~~~~~~~~
| compilation terminated.
| [2/11] /home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/recipe-sysroot-native/usr/bin/arm-phytec-linux-gnueabi/arm-phytec-linux-gnueabi-gcc -DAWS_SDKUTILS_EXPORTS -DAWS_SDKUTILS_USE_IMPORT_EXPORT -DHAVE_SYSCONF -Daws_c_sdkutils_EXPORTS -I/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/git/include -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0=/usr/src/debug/aws-c-sdkutils/0.1.11-r0                      -fdebug-prefix-map=/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0=/usr/src/debug/aws-c-sdkutils/0.1.11-r0                      -fdebug-prefix-map=/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/rqm/projects/yocto_hardknott/build/tmp/work/cortexa9t2hf-neon-phytec-linux-gnueabi/aws-c-sdkutils/0.1.11-r0/r [... cut ...]
ERROR: Task (/home/rqm/projects/yocto_hardknott/sources/poky/../meta-aws/recipes-sdk/aws-c-sdkutils/aws-c-sdkutils_0.1.11.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3682 tasks of which 3681 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 3 seconds

Summary: 1 task failed:
  /home/rqm/projects/yocto_hardknott/sources/poky/../meta-aws/recipes-sdk/aws-c-sdkutils/aws-c-sdkutils_0.1.11.bb:do_compile
Summary: There were 3 WARNING messages.
Summary: There was 1 ERROR message, returning a non-zero exit code.`

thomas-roos, 2023-07-03:
Hi, tried to reproduce this error, but works just fine.
Might the issue is in the meta-ksp0671 ?
Look here what I've tried: https://github.com/aws4embeddedlinux/meta-aws-demos/tree/master-next
Might you can provide a minimal expample to reproduce this issue based on this?

eflavio, 2023-07-06:
Hi Thomas,
Thanks for checking. Indeed, the problem lied in one of the other layers, namely meta-iot-cloud. It also contains recipes related to AWS and had higher layer priority than meta-aws.

I lowered the layer priority of meta-iot-cloud to 5 and everything works fine. I achieved this by adding this line to my local.conf file:
`BBFILE_PRIORITY_iot-cloud = "5"`

closed 2023-07-06 by eflavio