aws4embeddedlinux/meta-aws #5583: aws-crt-cpp-config.cmake from the crt/aws-crt-cpp/cmake is not included in yocto build

AbhayPatil, 2023-09-29:
I want to build C++ S3Client using yocto. I am trying to build aws-sdk-cpp using meta-aws. I followed following steps -

1. I copied the [aws-sdk-cpp_1.11.162.bb](https://github.com/aws4embeddedlinux/meta-aws/blob/master/recipes-sdk/aws-sdk-cpp/aws-sdk-cpp_1.11.162.bb) into my recipes directory.
2. Ran the following command - `$ MACHINE=machine-name bitbake aws-sdk-cpp`

I encountered following error -
```
| CMake Error at cmake/AwsFindPackage.cmake:19 (find_package):
|   By not providing "Findaws-crt-cpp.cmake" in CMAKE_MODULE_PATH this project
|   has asked CMake to find a package configuration file provided by
|   "aws-crt-cpp", but CMake did not find one.
|
|   Could not find a package configuration file provided by "aws-crt-cpp" with
|   any of the following names:
|
|     aws-crt-cppConfig.cmake
|     aws-crt-cpp-config.cmake
|
|   Add the installation prefix of "aws-crt-cpp" to CMAKE_PREFIX_PATH or set
|   "aws-crt-cpp_DIR" to a directory containing one of the above files.  If
|   "aws-crt-cpp" provides a separate development package or SDK, be sure it
|   has been installed.
| Call Stack (most recent call first):
|   src/aws-cpp-sdk-core/CMakeLists.txt:586 (aws_use_package)
```
I also tried to change the `SRC_URI` from `git://github.com/aws/aws-sdk-cpp.git;protocol=https;branch=main \` to     `gitsm://github.com/aws/aws-sdk-cpp.git;protocol=https;branch=main \` to account for submodules but that did not help and I got the same error message again.

Another thing I tried was as per the suggestion in error message - 
```
EXTRA_OECMAKE += "\
    -DCMAKE_PREFIX_PATH=crt/aws_crt_cpp/cmake \
"
```
But this also did not work.

Please let me know how to build aws-sdk-cpp using Yocto.
Please let me know if you need anything for this.

AbhayPatil, 2023-09-29:
Please ignore this issue for now. I think the above way of using recipe files is wrong.

thomas-roos, 2023-09-29:
Hi,
you need to include the whole meta-aws layer in your bblayers.conf not just copy one recipe. If you copy need to copy ALL dependent recipes as well but will lack the comfort of easy upgrades (https://docs.yoctoproject.org/overview-manual/concepts.html#layers).
Let me know if you still need help.
Will close this issue now.

closed 2023-09-29 by thomas-roos