aws4embeddedlinux/meta-aws #7265: ModuleNotFoundError: No module named 'botocore.vendored.six.moves'

landgraf, 2024-02-01:
Failed to build in master.

```
Build Configuration:
BB_VERSION           = "2.7.2"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "x86_64-poky-linux"
MACHINE              = "qemux86-64"
DISTRO               = "poky"
DISTRO_VERSION       = "4.3+snapshot-d50c4a1e362d310ff231adb7366712b70b668a1a"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       
meta-virtualization  = "master:8000c8d073b91fe18e41012204b969f6afd7ea89"
meta-filesystems     
meta-gnome           
meta-initramfs       
meta-multimedia      
meta-networking      
meta-oe              
meta-perl            
meta-python          
meta-webserver       
meta-xfce            = "<unknown>:<unknown>"
meta-aws             = "master:2aa42f166ef4d61fe4dc30638b4ca23c875a9682"
```

```
o/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/temp/run.do_compile.2679383', 1, None, None)
ERROR: Logfile of failure stored in: /mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/temp/log.do_compile.2679383
Log data follows:
| DEBUG: Executing shell function do_compile
| * Getting build dependencies for wheel...
| * Building wheel...
| Traceback (most recent call last):
|   File "/mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/recipe-sysroot-native/usr/lib/python3.12/site-packages/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
|     main()
|   File "/mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/recipe-sysroot-native/usr/lib/python3.12/site-packages/pyproject_hooks/_in_process/_in_process.py", line 335, in main
|     json_out['return_val'] = hook(**hook_input['kwargs'])
|                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
|   File "/mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/recipe-sysroot-native/usr/lib/python3.12/site-packages/pyproject_hooks/_in_process/_in_process.py", line 251, in build_wheel
|     return _build_backend().build_wheel(wheel_directory, config_settings,
|            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
|   File "/mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/git/backends/pep517.py", line 64, in build_wheel
|     _inject_wheel_extras(os.path.join(wheel_directory, whl_filename))
|   File "/mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/git/backends/pep517.py", line 189, in _inject_wheel_extras
|     _build_and_inject_ac_index(BUILD_DIR, extracted_wheel_dir)
|   File "/mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/git/backends/pep517.py", line 217, in _build_and_inject_ac_index
|     ac_index_build_name = _build_ac_index(build_dir)
|                           ^^^^^^^^^^^^^^^^^^^^^^^^^^
|   File "/mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/git/backends/pep517.py", line 228, in _build_ac_index
|     from awscli.autocomplete.generator import generate_index
|   File "/mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/git/awscli/autocomplete/generator.py", line 17, in <module>
|     from awscli.autocomplete.serverside.indexer import APICallIndexer
|   File "/mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/git/awscli/autocomplete/serverside/indexer.py", line 18, in <module>
|     import awscli.clidriver
|   File "/mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/git/awscli/clidriver.py", line 21, in <module>
|     import botocore.session
|   File "/mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/git/awscli/botocore/session.py", line 27, in <module>
|     import botocore.client
|   File "/mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/git/awscli/botocore/client.py", line 16, in <module>
|     from botocore import UNSIGNED, waiter, xform_name
|   File "/mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/git/awscli/botocore/waiter.py", line 17, in <module>
|     from botocore.docs.docstring import WaiterDocstring
|   File "/mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/git/awscli/botocore/docs/__init__.py", line 15, in <module>
|     from botocore.docs.service import ServiceDocumenter
|   File "/mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/git/awscli/botocore/docs/service.py", line 13, in <module>
|     from botocore.docs.bcdoc.restdoc import DocumentStructure
|   File "/mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/git/awscli/botocore/docs/bcdoc/restdoc.py", line 15, in <module>
|     from botocore.compat import OrderedDict
|   File "/mnt/builds/smth/smthagent_ng/build/tmp/work/core2-64-poky-linux/aws-cli-v2/2.15.11/git/awscli/botocore/compat.py", line 36, in <module>
|     from botocore.vendored.six.moves import http_client
| ModuleNotFoundError: No module named 'botocore.vendored.six.moves'
```

thomas-roos, 2024-02-02:
Hi, thank you for reporting. Will look into this issue and come back here.

thomas-roos, 2024-02-05:
This is the commit breaking it: https://github.com/yoctoproject/poky/commit/78e30d940d1a931c65a14d864494d92f0889035e
-> further investigation needs to be done how this could be fixed.

thomas-roos, 2024-02-05:
Seems like python dropped distutils support fully, not sure what we can do here - proper way would be the aws-cli-v2 team is to support to be python 3.12.x
-> found this ticket here: https://github.com/aws/aws-cli/issues/8342

thomas-roos, 2024-02-06:
Will ask internally if there is an estimated time to support python 3.12.x

thomas-roos, 2024-02-13:
No estimated time yet.
this is the right ticket -> https://github.com/aws/aws-cli/issues/8342

thomas-roos, 2024-02-26:
By the way on nanbield this is working fine, since this does not have python 3.12.x

ncopa, 2024-03-20:
I am in the process of upgrading python to 3.12 on Alpine Linux and this is affecting us also.

papoteur-mga, 2024-04-17:
In Mageia Cauldron, we have already Python 3.12, but aws-cli can't be re build because of this issue.

thomas-roos, 2024-09-23:
Python 3.12 is now supported. Recipe reactivated and merged.

PR #9681 (2024-09-23): aws-cli-v2: re-enable, python 3.12 compatible
Closes #7265

*Issue #, if available:*

*Description of changes:*

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
commit ea51b146 (2024-09-23): aws-cli-v2: re-enable, python 3.12 compatible

Closes #7265
closed 2024-09-23 by thomas-roos