aws4embeddedlinux/meta-aws #8492: Couldn’t able to provision GreenGrass Core Device in Yocto Jetson Images

manojrampur, 2024-05-14:
Hi Team,

We’re currently working on building Yocto Images for **Jetson AGX Xavier Developer Kit** and **Jetson TX2 Developer Kit** devices. We’ve successfully built two different images for AWS, but we’re encountering issues with provisioning the Greengrass Core Device in AWS Image.

Here are more details about the Images:

**Yocto AWS Greengrass Core Image:**

Build:
We’re cloning the kirkstone branch meta-aws layer. 
Commit ID: 1309ada6690cf59d6d36c77808b4d698da997259

Testing on Device:
When trying to provision the Greengrass core device on the device, we’re unable to do so.

> Note: We’ve successfully built an image for Raspberry Pi devices, and provisioning works well on the device. We’ve flashed Ubuntu on Jetson Devices and provisioning works, but not through Yocto OS.

Can you please confirm whether the above-mentioned Jetson devices support AWS Greengrass Core Device provisioning for Yocto build? 
Also, could you please confirm whether Yocto support is still available for the above-mentioned devices?

thomas-roos, 2024-05-14:
Hi, can you please give more details what is not working? Error message etc?
"We’ve successfully built an image for Raspberry Pi devices" -> did you build this with yocto?
As far as we know there should not be a limitation with devices.

thomas-roos, 2024-05-21:
any updates on this?

thomas-roos, 2024-06-14:
Closing because of missing information.

closed 2024-06-14 by thomas-roos