aws4embeddedlinux/meta-aws #8536: Yocto Kirkstone, awscrt and awsiotsdk both install with version number 1.0.0.dev0

akarnil, 2024-05-17:
I am building a small Yocto image that comes with all of the necessary components for a Greengrass device, I am using the latest Kirkstone branch, my image install is `IMAGE_INSTALL += " greengrass-bin aws-iot-device-sdk-python-v2"`

I do get `awscrt` and `awsiotsdk` installed onto the target device but both show as 1.0.0.dev0 instead of their actual versions (0.20.9 and 1.21.4)  
```
root@raspberrypi4:~# pip3 show awscrt
Name: awscrt
Version: 1.0.0.dev0
Summary: A common runtime for AWS Python projects
Home-page: https://github.com/awslabs/aws-crt-python
Author: Amazon Web Services, Inc
Author-email: aws-sdk-common-runtime@amazon.com
License: Apache 2.0
Location: /usr/lib/python3.10/site-packages
Requires: 
Required-by: awsiotsdk
root@raspberrypi4:~# pip3 show awsiotsdk
Name: awsiotsdk
Version: 1.0.0.dev0
Summary: AWS IoT SDK based on the AWS Common Runtime
Home-page: https://github.com/aws/aws-iot-device-sdk-python-v2
Author: AWS SDK Common Runtime Team
Author-email: 
License: License :: OSI Approved :: Apache Software License
Location: /usr/lib/python3.10/site-packages
Requires: awscrt
Required-by: 
```

Is it possible to get this fixed? the issue is that a component I am working with which I do not have access to the source of, one of the preliminary steps it takes is to install awsiotc through pip, it sees that 1.0.0.dev0 exists but it will still try to get the latest version which does build since cmake is missing from the target.

Thank you

thomas-roos, 2024-05-17:
Will have a look into this.

thomas-roos, 2024-05-17:
This is the reason - let's see if I can add this fix to the recipe.

https://github.com/aws/aws-iot-device-sdk-python-v2/blob/83c323bc1cdf7b84fc9405c8c9fa39e2b297408e/documents/FAQ.md?plain=1#L49

akarnil, 2024-05-17:
Let me know and I will be happy to test

thomas-roos, 2024-05-17:
Would be great if you could test this: https://github.com/aws4embeddedlinux/meta-aws/pull/8538

akarnil, 2024-05-17:
Thank you @thomas-roos 

This fixes my issue

```
root@raspberrypi4:~# pip3 show awscrt
Name: awscrt
Version: 0.20.9
Summary: A common runtime for AWS Python projects
Home-page: https://github.com/awslabs/aws-crt-python
Author: Amazon Web Services, Inc
Author-email: aws-sdk-common-runtime@amazon.com
License: Apache 2.0
Location: /usr/lib/python3.10/site-packages
Requires: 
Required-by: awsiotsdk
root@raspberrypi4:~# pip3 show awsiotsdk
Name: awsiotsdk
Version: 1.21.4
Summary: AWS IoT SDK based on the AWS Common Runtime
Home-page: https://github.com/aws/aws-iot-device-sdk-python-v2
Author: AWS SDK Common Runtime Team
Author-email: 
License: License :: OSI Approved :: Apache Software License
Location: /usr/lib/python3.10/site-packages
Requires: awscrt
Required-by: 
```