aws4embeddedlinux/meta-aws #9442: botocore SRCREV are all invalid (not found on master branch)

antoine-sac, 2024-08-28:
Hello,

Sadly it looks like someone force-pushed on and made all the SRCREV for python3-botocore invalid. This seems to affect all botocore SRCREV across all versions.

For example, this is the current python3-botocore SRCREV and it is not attached to any branch.
https://github.com/boto/botocore/commit/894686f5fc8d1440715877ec361695fb71fc900c

This means yocto builds fail with :

```
ERROR: python3-botocore-1.34.47-r0 do_fetch: Fetcher failure: Unable to find revision 91985d79e71c71f581095abc8f1789bd2f6c156d in branch master even from upstream
```

(This particular commit was the fairly old one we were using).

thomas-roos, 2024-08-28:
Thank you for reporting. Will look into that next week after being back from vacation.

antoine-sac, 2024-08-29:
As a temporary fix, we forked the meta-aws repo and fixed the commit SRCREV for the kirkstone branch. Feel free to use it if it's helpful: https://github.com/hestiia-engineering/meta-aws/tree/kirkstone

Note the botocore devs are lookink into restoring the commit history so the underlying issue should fix itself soon.

ndivecha, 2024-08-29:
> As a temporary fix, we forked the meta-aws repo and fixed the commit SRCREV for the kirkstone branch. Feel free to use it if it's helpful: [...]

ah ha! much appreciated... I guess work continues :-) 

Cheers

thomas-roos, 2024-08-31:
botocore master branch should be fixed by mid of next week.

thomas-roos, 2024-09-04:
This is fixed now thankfully. https://github.com/boto/botocore/issues/3241

closed 2024-09-04 by thomas-roos