aws4embeddedlinux/meta-aws #9501: Build error in Greengrass with fleet provisioning on scarthgap branch

mpichette, 2024-09-05:
I was successfully able to do a Yocto build for the greengrass demo but when switching over to the normal greengrass recipe with fleet provisioning enabled I'm encountering the following error:

`patch: **** Can't create temporary file /workdir/poky/rpi/tmp/work/cortexa72-poky-linux/greengrass-bin/2.12.6/image/greengrass/v2/alts/init/distro/bin/loader.oPXAGai : No such file or directory`

The full contents of /workdir/poky/rpi/tmp/work/cortexa72-poky-linux/greengrass-bin/2.12.6/temp/log.do_install.455 below:

`DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['/workdir/poky/meta/recipes-core/base-files/base-files_3.0.14.bb:do_populate_sysroot', '/workdir/poky/meta/recipes-core/base-passwd/base-passwd_3.6.3.bb:do_populate_sysroot', '/workdir/poky/meta/recipes-core/glibc/glibc_2.39.bb:do_populate_sysroot', '/workdir/poky/meta/recipes-core/systemd/systemd-systemctl-native.bb:do_populate_sysroot', '/workdir/poky/meta/recipes-devtools/gcc/gcc-cross_13.3.bb:do_populate_sysroot', '/workdir/poky/meta/recipes-devtools/gcc/gcc-runtime_13.3.bb:do_populate_sysroot', '/workdir/poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', '/workdir/poky/meta/recipes-extended/shadow/shadow-sysroot_4.6.bb:do_populate_sysroot', '/workdir/poky/meta/recipes-extended/shadow/shadow_4.14.2.bb:do_populate_sysroot', 'virtual:native:/workdir/poky/meta/recipes-core/gettext/gettext_0.22.5.bb:do_populate_sysroot', 'virtual:native:/workdir/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/workdir/poky/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/workdir/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/workdir/poky/meta/recipes-extended/shadow/shadow_4.14.2.bb:do_populate_sysroot', 'virtual:native:/workdir/poky/meta/recipes-extended/unzip/unzip_6.0.bb:do_populate_sysroot']
NOTE: Installed into sysroot: ['quilt-native', 'patch-native']
NOTE: Skipping as already exists in sysroot: ['base-files', 'base-passwd', 'gettext-minimal-native', 'glibc', 'libxcrypt', 'systemd-systemctl-native', 'binutils-cross-aarch64', 'gcc-cross-aarch64', 'gcc-runtime', 'libgcc', 'libtool-native', 'm4-native', 'opkg-utils', 'shadow-sysroot', 'shadow', 'texinfo-dummy-native', 'linux-libc-headers', 'gettext-native', 'zlib-native', 'flex-native', 'gnu-config-native', 'pkgconfig-native', 'pseudo-native', 'shadow-native', 'unzip-native', 'xz-native', 'zstd-native', 'attr-native', 'gmp-native', 'libbsd-native', 'libmd-native', 'libmpc-native', 'mpfr-native', 'sqlite3-native']
DEBUG: sed -e 's:^[^/]*/:/workdir/poky/rpi/tmp/work/cortexa72-poky-linux/greengrass-bin/2.12.6/recipe-sysroot-native/:g' /workdir/poky/rpi/tmp/sysroots-components/aarch64/quilt-native/fixmepath | xargs sed -i -e 's:FIXMESTAGINGDIRTARGET:/workdir/poky/rpi/tmp/work/cortexa72-poky-linux/greengrass-bin/2.12.6/recipe-sysroot:g; s:FIXMESTAGINGDIRHOST:/workdir/poky/rpi/tmp/work/cortexa72-poky-linux/greengrass-bin/2.12.6/recipe-sysroot-native:g' -e 's:FIXME_PSEUDO_SYSROOT:/workdir/poky/rpi/tmp/sysroots-components/aarch64/pseudo-native:g' -e 's:FIXME_HOSTTOOLS_DIR:/workdir/poky/rpi/tmp/hosttools:g' -e 's:FIXME_PKGDATA_DIR:/workdir/poky/rpi/tmp/pkgdata/raspberrypi4-64:g' -e 's:FIXME_PSEUDO_LOCALSTATEDIR:/workdir/poky/rpi/tmp/work/cortexa72-poky-linux/greengrass-bin/2.12.6/pseudo/:g' -e 's:FIXME_LOGFIFO:/workdir/poky/rpi/tmp/work/cortexa72-poky-linux/greengrass-bin/2.12.6/temp/fifo.455:g'
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing shell function do_install
WARNING: exit code 2 from a shell command.
patch: **** Can't create temporary file /workdir/poky/rpi/tmp/work/cortexa72-poky-linux/greengrass-bin/2.12.6/image/greengrass/v2/alts/init/distro/bin/loader.oPXAGai : No such file or directory
`

`pokyuser@b87f06294034:/workdir/poky/rpi$ ls -al /workdir/poky/rpi/tmp/work/cortexa72-poky-linux/greengrass-bin/2.12.6/image/greengrass/v2/alts/init
total 12
drwxr-xr-x 2 pokyuser pokyuser 4096 Sep  4 17:04 .
drwxr-xr-x 3 pokyuser pokyuser 4096 Sep  4 17:04 ..
lrwxrwxrwx 1 pokyuser pokyuser   97 Sep  4 17:04 distro -> /greengrass/v2/packages/artifacts-unarchived/aws.greengrass.Nucleus/2.12.6/aws.greengrass.nucleus
pokyuser@b87f06294034:/workdir/poky/rpi$ ls -al /workdir/poky/rpi/tmp/work/cortexa72-poky-linux/greengrass-bin/2.12.6/image/greengrass/v2/alts/init/distro
lrwxrwxrwx 1 pokyuser pokyuser 97 Sep  4 17:04 /workdir/poky/rpi/tmp/work/cortexa72-poky-linux/greengrass-bin/2.12.6/image/greengrass/v2/alts/init/distro -> /greengrass/v2/packages/artifacts-unarchived/aws.greengrass.Nucleus/2.12.6/aws.greengrass.nucleus`

thomas-roos, 2024-09-05:
Thank you for reporting. Will try reproduce.

thomas-roos, 2024-09-06:
Hi, can you please provide more information. Minimal example? Layers used (commit sha). Machine you are building for.  Ty!

mpichette, 2024-09-06:
**bblayers**

`# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /workdir/poky/meta \
  /workdir/poky/meta-poky \
  /workdir/poky/meta-yocto-bsp \
  /workdir/poky/meta-raspberrypi \
  /workdir/poky/meta-openembedded/meta-oe \
  /workdir/poky/meta-openembedded/meta-python \
  /workdir/poky/meta-openembedded/meta-multimedia \
  /workdir/poky/meta-openembedded/meta-networking \
  /workdir/poky/meta-java \
  /workdir/poky/meta-aws \
  "
`

**local.conf**

MACHINE ?= "raspberrypi4-64"
INIT_MANAGER = "systemd"
INHERIT += "cve-check"

include cve-extra-exclusions.inc

LICENSE_FLAGS_ACCEPTED = "synaptics-killswitch"

DISTRO_FEATURES += "systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = ""

IMAGE_INSTALL:append = " greengrass-bin"
PACKAGECONFIG:pn-greengrass-bin = "fleetprovisioning"

GGV2_DATA_EP     = "REDACTED-ats.iot.us-east-1.amazonaws.com"
GGV2_CRED_EP     = "REDACTED.credentials.iot.us-east-1.amazonaws.com"
GGV2_REGION      = "us-east-1"
GGV2_THING_NAME  = "REDACTED"
GGV2_TES_RALIAS  = "REDACTED"

# Fleet Provisioning
GGV2_THING_GROUP = "REDACTED"
GGV2_FLEET_PROVISIONING_TEMPLATE_NAME = "REDACTED"
#GGV2_FLEETPROVISIONING_VERSION = "2.12.6"

# Raspberry Pi Options

ENABLE_DWC2_HOST = "1"
RPI_EXTRA_CONFIG="\n\
otg_mode=1\n\
"

HDMI_GROUP = "2"
HDMI_MODE = "87"
CONFIG_HDMI_BOOST = "10"
HDMI_CVT = "1024 600 60 6 0 0 0"
HDMI_FORCE_HOTPLUG = "1"

Commit SHA for meta-aws
commit 610c9a8ebee823af833ccd48dcc7462dd3bed1a2 (HEAD -> scarthgap, origin/scarthgap)

Commit SHA for base poky/yocto
commit f43f393ef0246b7bee6eed8bcf8271cf2b8cdf40 (HEAD -> my-scarthgap, origin/scarthgap)

I'm building for a Raspberry Pi CM4 on a Mac (M3) in a Poky container.

thomas-roos, 2024-09-09:
Was able to reproduce.
Please try if this is fixing the issue on your side. 
https://github.com/aws4embeddedlinux/meta-aws/pull/9523/commits/df732be76140521b8af8390a754b3d8d0b1aa893

thomas-roos, 2024-09-10:
And reopen if there is still an issue. Ty!

PR #9523 (2024-09-09): greengrass-bin: fix fleetprovisioning
new version of greengrass-bin changed path of loader, patch failed. Also update sha of aws.greengrass.FleetProvisioningByClaim.jar latest.

fix #9501

*Issue #, if available:*

*Description of changes:*

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
commit 521e9fc8 (2024-09-09): greengrass-bin: fix fleetprovisioning

new version of greengrass-bin changed path of loader, patch failed.
Also update sha of aws.greengrass.FleetProvisioningByClaim.jar latest.

fix #9501
commit 00c79270 (2024-09-09): greengrass-bin: fix fleetprovisioning

new version of greengrass-bin changed path of loader, patch failed.
Also update sha of aws.greengrass.FleetProvisioningByClaim.jar latest.

fix #9501
commit 096ca6e9 (2024-09-09): greengrass-bin: fix fleetprovisioning

new version of greengrass-bin changed path of loader, patch failed.
Also update sha of aws.greengrass.FleetProvisioningByClaim.jar latest.

fix #9501
closed 2024-09-10 by thomas-roos