aws4embeddedlinux/meta-aws #9647: AWS IoT SDK and boto3 broken on mickledore

vaigen, 2024-09-18:
Hi there,

I know mickledore is not actively maintained, but it seems completely broken. 

```
```

thomas-roos, 2024-09-22:
Hi, thanks for reporting. Can you please share your other layers used in your configuration. May another layer is providing a wrong version.
Will try to reproduce the issue. Not aware of it.

thomas-roos, 2024-09-23:
Need more information reproduce - on my side this seems to work:
```
root@qemux86-64:~# python3
Python 3.11.5 (main, Aug 24 2023, 12:23:19) [GCC 12.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import boto3
>>> 
─────────────
```

vaigen, 2024-09-23:
Hi Thomas, I've modified the layers to build meta-aws as the last layer, still see the same issue. I see you are testing on qemu x86. The problem I reported is on an aarch64 target. 

It is an NXP target using https://github.com/nxp-imx/meta-imx/blob/nanbield-6.6.3-1.0.0/tools/imx-setup-release.sh#L167-L178.

thomas-roos, 2024-09-24:
Hi,
last layer does not mean anything without changing layer priority. I still think there is some recipe in the other layers that disturbing meta-aws. And this should be BBMASKED (eg. https://github.com/aws4embeddedlinux/meta-aws/blob/fe39a4c0930199e29b4813eab45323ff6a4959aa/conf/layer.conf#L35)
Tested with qemuarm64:
```
Poky (Yocto Project Reference Distro) 4.2.4 qemuarm64 /dev/ttyAMA0

qemuarm64 login: root
root@qemuarm64:~# python3
Python 3.11.5 (main, Aug 24 2023, 12:23:19) [GCC 12.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import boto3

```

vaigen, 2024-09-24:
I meant priority yes. But true, meta-virtualization does have some conflicting versions. Thanks for pointing that out. Let me override those and try again. Appreciate your help!

thomas-roos, 2024-09-24:
If you found the bad recipe it would be great to add this to the mask here. Patches welcome 😉

vaigen, 2024-09-25:
Ok, so I bumped up meta-aws priority to the highest there is on our stack and masked a bunch of boto* recipes from meta-python-ai. The problem still exists.

```
aws --version
Fatal error condition occurred in /usr/src/debug/aws-c-common/0.9.8-r0/source/allocator.c:187: num != 0 && size != 0
Exiting Application
################################################################################
Stack trace:
################################################################################
/usr/lib/libaws-c-common.so.1(aws_backtrace_print+0x68) [0xffffbba9fd58]
/usr/lib/libaws-c-common.so.1(aws_fatal_assert+0x58) [0xffffbba832c8]
/usr/lib/libaws-c-common.so.1(aws_mem_calloc+0xe4) [0xffffbba7ed64]
/usr/lib/libaws-c-s3.so.0unstable(aws_s3_platform_info_loader_new+0x78) [0xffffbbd7c288]
/usr/lib/libaws-c-s3.so.0unstable(aws_s3_library_init+0x60) [0xffffbbd6c970]
/usr/lib/python3.11/site-packages/_awscrt.abi3.so(PyInit__awscrt+0x118) [0xffffbbdc7488]
/usr/lib/libpython3.11.so.1.0(+0x252a44) [0xffffbd7f2a44]
/usr/lib/libpython3.11.so.1.0(+0x24f7c0) [0xffffbd7ef7c0]
/usr/lib/libpython3.11.so.1.0(+0x193e44) [0xffffbd733e44]
```

The versions of python packages installed match the ones from meta-aws too.

```
pip3 list
Package           Version
----------------- --------------
argh              0.26.2
attrs             22.2.0
awscli            1.29.79
awscrt            1.0.0.dev0
awsiotsdk         1.0.0.dev0
beniget           0.4.1
boto3             1.28.83
botocore          1.31.83
```

I guess the next step is to strip layer by layer and see what gives.

thomas-roos, 2024-09-25:
I guess you are using latest mickledore commit?
(https://github.com/aws4embeddedlinux/meta-aws/commit/fe39a4c0930199e29b4813eab45323ff6a4959aa)
Otherwise yes, strip layer by layer and hopefully at one point you know why this is failing.
I'm curious as well!

thomas-roos, 2024-10-07:
Is there any updates on this? Can it be closed?

thomas-roos, 2024-10-14:
Closing. Please open if the issue still exists.

closed 2024-10-14 by thomas-roos