aws4embeddedlinux/meta-aws #9745: BUILD_ONLY no longer works

parsley72, 2024-09-28:
We've been using this layer for AWS S3 support. Originally we used aws-sdk-cpp 1.9.379 (Nov 6, 2022). We only needed S3 support so we added recipes-devtools/aws-sdk-cpp/aws-sdk-cpp_1.%.bbappend:
```
LIBS_BUILD_ON_ARM32 = "-DBUILD_ONLY='s3'"
```
and this worked fine. But recently I tried to upgrade to the latest release aws-sdk-cpp 1.11.404 (Sep 13, 2024) and found the resulting image couldn't be installed on our device - it's too large. After changing the checkout a few times and rebuilding I found that this commit from @thomas-roos hugely increases the size of the library:
https://github.com/aws4embeddedlinux/meta-aws/commit/2ec57954e61025dd9c676c6e01434dc4344a86d9
```
ptest: run integration tests
oelint-adv: no errors, no warnings
all libs: now all libs a build by default, separate packages, aws-sdk-cpp package contains all libs
```
How can we continue to build only the S3 support?

parsley72, 2024-09-29:
This seems to work if I change .bbappend to:
```
EXTRA_OECMAKE += "-DBUILD_ONLY='s3'"
```

thomas-roos, 2024-09-30:
Hi, 
this LIBS_BUILD_ON_ARM32 was just used to restrict libs to build on arm32, some failed. So this is not the case anymore thus this variable is removed.
Your suggestion is correct, also like to point out that the lib is split into different packages, so e.g. aws-cpp-sdk-s3 can be installed instead of all as well. But still all libs will be build, which take some time.

closed 2024-09-30 by thomas-roos