aws4embeddedlinux/meta-aws #12418: Patch/build failure in greengrass-bin on Scarthgap branch

mpichette, 2025-05-06:
I'm running into an issue building the greengrass-bin recipe on Scarthgap. It's complaining while applying the greengrass.service.diff patch. I think removing the `do_patch[no_exec] = "1" i`n this commit: #0eac711483da1017b5fbaf159b7e35faa44e9788 is causing the patch to be applied prior to the install where the greengrass.service file is available. It looks like the patch was normally executed manually here: https://github.com/aws4embeddedlinux/meta-aws/blob/8409c1b74722964befed5af53d750b0e3a48dcd4/recipes-iot/aws-iot-greengrass/greengrass-bin_2.14.0.bb#L107

Error logs below:

`DEBUG: Using /home/mpichette/Repos/workdir/sources/meta-scralf/recipes-iot/aws-iot-greengrass/files/claim.root.pem for claim.root.pem
NOTE: Applying patch '001-serviced-startup-sh.patch' (../sources/meta-aws/recipes-iot/aws-iot-greengrass/greengrass-bin/001-serviced-startup-sh.patch)
NOTE: Applying patch 'greengrass.service.diff' (../sources/meta-aws/recipes-iot/aws-iot-greengrass/greengrass-bin/greengrass.service.diff)
ERROR: Applying patch 'greengrass.service.diff' on target directory '/home/mpichette/Repos/workdir/build-raspberrypi-armv8/tmp/work/cortexa53-scralf-linux/greengrass-bin/2.14.0'
CmdError('quilt --quiltrc /home/mpichette/Repos/workdir/build-raspberrypi-armv8/tmp/work/cortexa53-scralf-linux/greengrass-bin/2.14.0/recipe-sysroot-native/etc/quiltrc push', 0, "stdout: Applying patch greengrass.service.diff
can't find file to patch at input line 3
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- greengrass.service
|+++ greengrass.service
--------------------------
No file to patch.  Skipping patch.
patch unexpectedly ends in middle of line
1 out of 1 hunk ignored
patch unexpectedly ends in middle of line
Patch greengrass.service.diff does not apply (enforce with -f)

stderr: ")
DEBUG: Python function patch_do_patch finished
DEBUG: Python function do_patch finished`

thomas-roos, 2025-05-07:
Thanks for reporting. The issue should be fixed now.

closed 2025-05-07 by thomas-roos

mpichette, 2025-05-07:
Thanks Thomas!

PR #12420 (2025-05-07): greengrass-bin: upgrade 2.14.0 -> 2.14.1
Fix #12418 also

*Issue #, if available:*

*Description of changes:*

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
PR #12419 (2025-05-07): greengrass-bin: upgrade 2.14.0 -> 2.14.1
Fix #12418 also

*Issue #, if available:*

*Description of changes:*

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
commit a0b333fc (2025-05-07): greengrass-bin: upgrade 2.14.0 -> 2.14.1

Fix #12418 also
commit de554d12 (2025-05-07): greengrass-bin: upgrade 2.14.0 -> 2.14.1

Fix #12418 also