aws4embeddedlinux/meta-aws #12832: Recipe aws-iot-device-sdk-embedded-c does not copy headers when populate_sdk with Yocto

rafamontano, 2025-06-25:
Hi,

When populate_sdk the headers are not being copied to the sysroot, which is required to cross-compile the application before including it into the image.

https://github.com/aws4embeddedlinux/meta-aws/blob/5cd6cb8c864939f3880fd1f1f51232b4ec588b89/recipes-sdk/aws-iot-device-sdk-embedded-c/aws-iot-device-sdk-embedded-c_202412.00.bb#L39

I have tried by removing do_install and using the CMakeLists to copy those files, it works and copies the headers, but the bitbake fails.

rafamontano, 2025-06-27:
Hi,

Thank you for your quick response and for discussing this internally!

You are correct – by "application," I mean the C/C++ code I'm writing on my host computer that will ultimately be deployed and run on the IMX8 device. This application needs to link against the SDK's libraries and utilize the corresponding header files for the TLS MQTT communication with AWS IoT Core and job execution functionalities.

Regarding your suggestion: Yes, having separate recipes for all contained libraries (and their respective headers) that are in the SDK would absolutely help me. This sounds like a much cleaner and more maintainable approach than trying to retroactively inject headers into an existing SDK recipe, or creating a single monolithic recipe just for all SDK components.

It would allow me to specifically pull in only the necessary dependencies for development on the host, without needing the full, device-specific image. This would greatly streamline my development workflow.

Thank you for considering this!

rafamontano, 2025-07-04:
Hi,

This approach seems to be working. I did not mention earlier that I'm working on Scarthgap, so I made some minor changes to the branch and optimized it for Scarthgap. I'm currently working with these libraries:

- core_mqtt
- jobs
- backoffalgorithm
- core_json

But it would be great to include all the libraries in the SDK. 

https://github.com/rafamontano/meta-aws/commit/15a6f6825c70a0a930a6e103966eed3f4e8e8611

Let me know if this makes sense to you.

Rafael

thomas-roos, 2025-07-04:
Hi,
looks good - yes, first master and then backport to non EOL versions of Yocto.
Hope I can find some time next week to work on this.

thomas-roos, 2025-07-14:
Hi,
did merge https://github.com/aws4embeddedlinux/meta-aws/commit/189e82520e7d5d50e032203a3941300973e1a63b - not in the release branch yet, but that should be it. Please provide feedback and feel free to close this if you are happy with it.
Latest next week I will ff merge scarthgap-next in scarthgap branch.
Cheers, Thomas

closed 2025-07-22 by thomas-roos