aws4embeddedlinux/meta-aws #13233: nativesdk-aws-crt-python build failure

ChenQi1989, 2025-08-08:
The recent changes/upgrades for aws-crt-python introduces a regression, that is, nativesdk-crt-python fails to build.

CMake Error at /PATH/TO/build-poky/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-aws-crt-python/0.27.4/recipe-sysroot-native/usr/share/cmake-4.0/Modules/FindPack\
ageHandleStandardArgs.cmake:227 (message):
  Could NOT find crypto (missing: crypto_LIBRARY crypto_INCLUDE_DIR)
Call Stack (most recent call first):
  /PATH/TO/build-poky/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-aws-crt-python/0.27.4/recipe-sysroot-native/usr/share/cmake-4.0/Modules/FindPackageHandleStan\
dardArgs.cmake:591 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/Findcrypto.cmake:75 (find_package_handle_standard_args)
  CMakeLists.txt:271 (find_package)

-- Configuring incomplete, errors occurred!
CMake Error at aws-c-common/cmake/AwsPrebuildDependency.cmake:65 (message):
  Configuration failed for dependency project S2N
Call Stack (most recent call first):
  CMakeLists.txt:77 (aws_prebuild_dependency)

thomas-roos, 2025-08-08:
which branch are you building?

ChenQi1989, 2025-08-08:
Both master and walnascar.

PR #13234 (2025-08-11): aws-crt-python: fix nativesdk build
That also needs explicit set of OpenSSL paths.

Closes: #13233

*Issue #, if available:*

*Description of changes:*

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
commit b9cc45a7 (2025-08-11): aws-crt-python: fix nativesdk build

That also needs explicit set of OpenSSL paths.

Closes: #13233
commit e521e9d3 (2025-08-11): aws-crt-python: fix nativesdk build

That also needs explicit set of OpenSSL paths.

Closes: #13233
(cherry picked from commit b9cc45a7a9d1a5a2fc71c340ccbe58612995d4a1)
commit bd0df5ff (2025-08-11): aws-crt-python: fix nativesdk build

That also needs explicit set of OpenSSL paths.

Closes: #13233
(cherry picked from commit b9cc45a7a9d1a5a2fc71c340ccbe58612995d4a1)